Architeuthis-Flux/Jumperless #1: Which breadboard?

atlaste, 2023-04-01:
I actually like the idea of having a prototype board that I can use for fast prototypes. 

I was wondering what breadboard you're using? They all seem to be made with sticky tape at the bottom which holds the contacts in place, so that makes it impossible to solder them right on to the PCB. Do you peel off the sticky tape with isopropyl?

Architeuthis-Flux, 2023-04-01:
After reading through a lot of heated debates about which breadboard is the best, I settled on the [Jameco Valuepro WBU-301-R](https://www.jameco.com/z/WBU-301-R-Jameco-ValuePro-400-Point-Solderless-Breadboard-3-3-Lx2-1-W_20601.html). They seem to hold up pretty well and have quite a few holes in the bottom to put screws into. I usually put a bit of epoxy in each hole so the threads on the [M1.7 8mm self tapping screws](https://www.amazon.com/Tapping-Phillips-Metric-Assortment-Stainless/dp/B07XHR75SX/ref=asc_df_B07XHR75SX/?tag=hyprod-20&linkCode=df0&hvadid=632022073146&hvpos=&hvnetw=g&hvrand=17875811370743167852&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9030773&hvtargid=pla-1928640504673&psc=1) can bite better. 

The trick to getting the sticky back off is to get a brand new one, and peel it off as soon as you open it. At the factory they don't really press the tape on very well so if you never put any pressure on it, it peels right off. 

If some is left I'll scrape it off with an xacto knife or tweezers dry, I experimented with isopropyl and it just makes a mess and doesn't really get it off. Sandpaper on the bottom of the board works too (I do that anyway right before I apply the [Z tape](https://shop.evilmadscientist.com/productsmenu/831) to make better contact.)

I'll write up a guide for sticking breadboards to PCBs. I've tested a crazy number of different ways but this one seems to work with minimal effort.

atlaste, 2023-04-04:
Yea I didn't quite understand how it was done. Seems like you don't solder the breadboard to the pcb? I'm just not sure how that part fits together. Or is it just screwed down?

I also don't quite understand how the routing itself works while checking the schematics. I have to admit, I'm mainly working on (different) ESP's, and they have roughly 45 pins. Quite a few of these are dedicated to certain tasks, which makes testing even more important. So that means that I would need 3 IC's per 8 pins to route 3x16=48 pins across a "bus" to the microcontroller. With 60 pins that's 8x3=24 IC's. That seems like quite a lot to put on a board.

HasanTheSyrian, 2023-11-06:
> After reading through a lot of heated debates about which breadboard is the best, I settled on the [Jameco Valuepro [...]

From what I understand you're using special clips, is that right?

Also, a master BOM that includes all the components in the bom.csv files + the other components like the breadboard would be nice. Maybe I can make one and add it in `/hardware`.