Architeuthis-Flux/Jumperless #7: Interface for voltage and current.

ericmoderbacher, 2023-07-02:
Maybe a little gui window to show the measured voltage and current continuously. Does not need to be part of another project or executable.

JamesNewton, 2023-07-03:
I've found graphs of current draw, over the last minute or even 10 seconds or so to be very very helpful. I got that on my latest bench power supply and I'm amazed at how often I use it. It's normally a "wait... what just happened?" sort of a thing, and then I look up and see "oh, something shorted" or "Oh... it's didn't pull any power for a second there, did the load come disconnected?" or things like that. So a little graph, along with the number would be super nice. V2.0 maybe.

ericmoderbacher, 2023-07-04:
@JamesNewton that's great input, as a software guy I don't really know what's useful to include.

Architeuthis-Flux, 2023-07-08:
@JamesNewton good idea, I use that little graph on my power supply all the time. At one point for testing I did it with printing a bunch of periods depending on the reading. But eventually I'm going to need to take the plunge and write a full on GUI. If you have any suggestions for which framework I should use, let me know.

JamesNewton, 2023-07-08:
@Architeuthis-Flux Oh, I have a very strong opinion on this one. HTML / SVG. Always write GUI's for the web. Then use e.g. Phonegap or Node or Electron or whatever to compile it down into a program. Always. Then you have a web interface, and program interfaces for every OS because all the work is done for browsers on every OS. And it's a standard that will never (knock on wood) go away. And... it's pretty easy to use, well documented, lots of tools to help make things easy. And... if you learn it, then you can apply it to building web sites.

Architeuthis-Flux, 2023-07-08:
@JamesNewton thanks for that, now that you say it, it does seem like that's totally the way to go considering that's how Wokwi is written and I'll probably need to include it in any GUI I make for this.

nilclass, 2023-12-04:
GUI aside, is there any way to get to those values currently? I see a [commented out `m` here](https://github.com/Architeuthis-Flux/Jumperless/blob/fb41f849f8834ca682aab47a836ab1fb83718420/JumperlessNano/src/main.cpp#L99), but it [doesn't do anything](https://github.com/Architeuthis-Flux/Jumperless/blob/fb41f849f8834ca682aab47a836ab1fb83718420/JumperlessNano/src/main.cpp#L134-L136)