Architeuthis-Flux/Jumperless #13: Not getting ground in breadboard

omeriko9, 2023-10-28:
Hey, just got my jumperless and assembled it per instructions, but I can't get GND in any of the breadboard pins except for the ground line.

As you can see, the leds are green signifying Ground, but the led won't turn on:

[image]

This is my Wokwi diagram but I also trying connecting row 0 to the Nano's GND, it still won't work:

[image]

When I connect the led directly to the Ground line (or using a jumper wire) I do get ground.

Any idea how can I fix/debug this?

Architeuthis-Flux, 2023-10-28:
Thanks for putting in this issue! I think I may have left some silly change in the code from debugging something else that isn't connecting Net 0 GND, there's a for loop somewhere starting from 1 instead of 0. I'll find it and put up a fix this morning.

Architeuthis-Flux, 2023-10-28:
Could you also let me know if you have Rev 2 or Rev 3? and if it works on any other rows? The 4 corners of the breadboard (row 1, 30, 31, 60) work slightly differently from the rest, so it's helpful to know whether the issue is just there. On my board, it seems to work elsewhere but not in the corners.

omeriko9, 2023-10-28:
Forgive my ignorance, how can I know which revision the board is? Here is the back of the board:
[image]

Architeuthis-Flux, 2023-10-28:
That's on me for not writing the revision number on the boards, you can tell by the location of the ADC (on Rev 3 it's in place of the Nano Disconnect solder jumpers). You have a Revision 2, which is good because that's all I have to test on right now and I found a fix for you. 

Right under the rail supply switch (top right looking at the back) there's a solder jumper that should be cut (if it's not, cut it). Run a jumper from the left side pad to the GND rail. Like this:
[image]
And you should be able to get GND on the 4 corners. 

I'm working on a more robust fix in firmware that will work on both revisions right now, so when I update the release you should update your firmware. 

[image]
The reason for this is the corners are considered a Special Function on Chip L and the pathfinding has to deal with them differently. It's kinda weird but it saved me from having to add another crosspoint switch to the board.

Architeuthis-Flux, 2023-10-28:
Let me know if that works for you.

omeriko9, 2023-10-28:
I will try and let you know, thanks for the elaborate explanation and the care you invest in this issue!

Architeuthis-Flux, 2023-10-28:
Hey, the bug should be fixed in the latest release. Just download firmware.uf2 from the releases page and it should work better. Still do the bodge though, it assumes you've done it. I'll keep hacking away to get it to work without the bodge, so I'll keep this open until that's fully done.

omeriko9, 2023-10-29:
Yes, the patch + latest firmware indeed fixed it, now I get GND on the edge rows. Nice :)

There's another issue where Wokwi + JumperlessWokwiBridge won't update the sketch on the Nano, do you want me to open a different issue for that or are you aware of that already?

Architeuthis-Flux, 2023-10-29:
You can open up an issue for that. But as of now, I haven't written the firmware to do that. It turns out it's pretty complex to grab all the libraries and compile everything into a hex file like that. It's completely feasible, just will take some work in a field of coding I'm not super good with.

For now, the best way to flash code is with a second USB cable, the Jumperless will reset the Arduino each time you change the wiring (I can change it if you don't want it to do that, or just cut the two solder jumpers labeled Reset.)

closed 2023-10-29 by Architeuthis-Flux