Architeuthis-Flux/Jumperless #15: CAD files for case?

nilclass, 2023-12-04:
@Architeuthis-Flux: I spotted this neat little case on one of the pictures in the README:
[image]

Any chance you could share some CAD files for it, so I can print one for myself? I can't find them anywhere..

nilclass, 2023-12-05:
Thanks so much for the quick reply :dash: 

I don't have a laser cutter, but I do have a 3d printer, so I'll try to print myself a box (when all you have is a :hammer: ...)

It will probably turn out horrible, given the amount of tiny features in the hair of the logo. Anyway, will give it a try :see_no_evil: .

The SVG you shared is a bit tricky to work with, since it has a lot of unconnected path segments. I tried to clean those up a bit, to get a shape that has a clearly defined "inside" and "outside" (which is required to turn it into a 3d model).

This is the result of that:
[image]

Architeuthis-Flux, 2023-12-15:
Hey, so I just got a bunch of cases in, and I already have your address so I'm gonna send you one. But it would be super cool to have a version of this case that is better suited to FDM so people can 3D print them at home. 

Here's the STL for the whole Jumperless so you have a shape to design around. The Nano headers didn't make it in there but they stick up exactly as high as the breadboard. 

[Jumperless.stl.zip](https://github.com/Architeuthis-Flux/Jumperless/files/13679454/Jumperless.stl.zip)

closed 2023-12-15 by Architeuthis-Flux

Architeuthis-Flux, 2023-12-17:
@nilclass Hey, your case has been sent out, it should be there sometime this week. Also, someone on the forums made a 3D printable stand (not really a case but it could be useful) as well. 

[1ZE15D330435626163](http://wwwapps.ups.com/WebTracking/processInputRequest?AgreeToTermsAndConditions=yes&loc=en_US&tracknum=1ZE15D330435626163)

[https://forum.jumperless.org/t/3d-printed-stand/57/3](https://forum.jumperless.org/t/3d-printed-stand/57/3)

nilclass, 2023-12-20:
@Architeuthis-Flux the case arrived, thanks so much, it's beautiful!

The stand looks useful too - not sure about it's height, but maybe I'll just cut off the bottom part before printing... I'll follow up on the forum :)