Architeuthis-Flux/Jumperless #20: Crosspoint Switch Logic

vinilsaji2003, 2024-01-02:
I would like to understand the logic behind the crosspoint switch,if we are using an 8*16 crosspoint switch array,then only 8 distinct connections can be made simultaneously-right,if we want more connections then nodes would be merged and the signals might get mixed up.How did you overcome this logical problem,I mean only 8 connections can be made simultaneously,but what if we need more connections from the same chip that might merge two or more nodes leading to signal mixing

Architeuthis-Flux, 2024-01-02:
Yeah that is by far the most complicated part of this whole project. I wrote a project log with some explanation on [Hackaday](https://hackaday.io/project/191238-jumperless/log/222530-the-code-part-2-pathfinding), but even that really isn't enough to explain exactly what's going on.

But basically, the crosspoints have 7 of their inputs connected to a some rows on the breadboard, and the remaining 17 connections are connected to the other 11 chips on the board (some have 2 going to the same chip). If it wants to make a connection to somewhere and the line is already being used for another connection, it looks for an intermediary chip that has an unused connection to both of the chips that want to be connected, and "hops" to where it needs to go through another chip. 

It stores what connections are made and what's available so it knows not to merge things that shouldn't be. Those are defined in [JumperlessNano/src/MatrixStateRP2040.cpp](https://github.com/Architeuthis-Flux/Jumperless/blob/main/JumperlessNano/src/MatrixStateRP2040.cpp) if you want to have a look at the state, the nitty gritty logic of the pathfinding, you can check that out in [JumperlessNano/src/NetsToChipConnections.cpp](https://github.com/Architeuthis-Flux/Jumperless/blob/main/JumperlessNano/src/NetsToChipConnections.cpp). 

If you want to see what's going on step-by-step, you can load this firmware onto a Raspberry Pi Pico and turn the Nets To Chip Connections debug flags on and it will give you a huge wall of data showing what's going on as it happens.

TL;DR It's complex as hell but it does work.

vinilsaji2003, 2024-01-02:
We are electronics engineering students and we would like to implement this project on our own to gain an experience in various domains involved in the project,our idea was to just implement the breadboard only,we had this idea going on in our head for the past 1.5 years,we use breadboards in our laboratories and at times it seemed very hard to connect with all the jumper wires and it was complex to troubleshoot,we had this idea of connecting each pin separately but it would need atleast 1770 connections as it stands,fortunately we found you,Though we were aware of the crosspoint switch ic's we are amused by how you have used it,would you be kind enough to tell us where we should go from hereon to implement this logic with deep understanding

Architeuthis-Flux, 2024-01-02:
Awesome, that sounds like a fun project. I would probably start with reading all the stuff on [breadWare](https://github.com/Architeuthis-Flux/breadWare), it's my earlier prototype for this and should be a bit simpler to follow. And It shows my thinking for a lot of the decisions I made on Jumperless. 

Some tips I could give you is not to try to do a full size breadboard at first, the number of connections and chips needed goes up really quickly, and 8x16 crosspoints just aren't big enough to handle that without even more tricks than I use here. 

Also, don't try to model a larger "fully connected" crossbar (like every switch has 8 connections to every other switch), you're better off making some assumptions about what people will realistically make on a breadboard and using fewer chips. 

Read up on [Telephone Exchanges](https://en.wikipedia.org/wiki/Telephone_exchange). Because that's basically what's going on with this, all the same ideas apply (the only difference is that you also want to minimize the number of switches a signal goes through because of the on resistance.) There are tons of really good papers from Bell Labs from the 60's, and just going on a Wikipedia rabbit-hole on these will help a lot. Start here and then go to the See Also section.
[Nonblocking Minimal Spanning Switch](https://en.wikipedia.org/wiki/Nonblocking_minimal_spanning_switch)
[Switching Circuit Theory](https://en.wikipedia.org/wiki/Switching_circuit_theory)

vinilsaji2003, 2024-01-02:
Thank you for replying,we are very grateful,we look forward to further mentorship from you

vinilsaji2003, 2024-01-05:
Could you please tell us in how many stages  you have implemented the clos network logic,is it 8 ic's for the first stage and then 4 for the upcoming stages,would we be able to make use of more connections if we implement it in more stages

Architeuthis-Flux, 2024-01-05:
I didn't really strictly implement any particular topology for this, but it's basically a 2 stage (just input and output layers) that can fall back to 3 stage if the 2 available paths are taken. The reason it had to be so different from those topologies is because in this case, the most important thing to worry about is the resistance of the switches themselves (~45 ohms each). 

The 4 special function chips are basically the same idea, except that they're wired "backwards" (16 pins to the outside world, 8 to the rest of the chips)

So, you absolutely can do one with more stages to get more nodes, just keep in mind that every switch you go through adds another 45 ohms.

Another trick I'm using to reduce the switch resistance is to power them with a lot more voltage than the datasheet writes as the maximum. Datasheet says Vdd - Vss is max 14.2V (which gives 65ohms on resistance, and I'm running them at Vdd - Vss ~18.5V to get that down to 45 ohms. 
I should test how much further I can push it, but so far, it works totally fine.

vinilsaji2003, 2024-01-05:
Ok,thankyou for sharing that vital piece of information

vinilsaji2003, 2024-01-05:
Sir,we have understood the logic used,thanks to your explanation,but when it comes to the schematic understanding of this project,it seems too complex to understand.so we would like to implement a miniature version of breadboard with may be only 5 or 10 columns,Would you be generous to give us any suggestions on how the connections should be made easily.one more doubt sir,when we try to implement only the breadboard without actually trying to import a nano into it would we require the control routing crosspoint switch ic's,forgive me if there is any misunderstanding,but i have found out that only the connections AB0,AB1,AC0,AC1.... etc are directly linked to the breadboard columns(am i right),i assume the rest were made to connect the nano in any configuration possible,Would i be bothering you if you could help us with a schematic when we want to make a breadboard with only 10 columns and without a nano connection just so as to ensure that our understanding is fully correct,as we are devoid of a lot of equipments and moral support from our college we are just keen to make sure that we dont make a lot of mistakes,also have you used any algorithm to connect AB0,AB1,AC0,AC1.....etc because they all seem to follow a specific pattern from chip to chip

vinilsaji2003, 2024-01-05:
Also could you explain how many columns from the breadboard is connected to AB0 alone,is it two, because there seems to be 28 AB0,AB1 .....etc like connections with an alternative path to another chip

Architeuthis-Flux, 2024-01-05:
The AB0 AB1... connections aren't connected anywhere externally, they're just connections between the chips, for example AB0 connects chip A to Chip B (lane 0). The connections that go to the breadboard columns are labeled as just numbers. 

Anyway, here's a much simpler schematic for 24 rows. It's fully interconnected so everything can connect to everything else without worrying about collisions, which should make the code much simpler. The 3 CH446Qs are controlled in parallel mode with an external microcontroller. So just put a header on the PCB and set the X0-X3 and Y0-Y2 for the address and then strobe the CS_A B or C high to make a connection. 
[image]

If you don't want to mess with the power supply, you can leave it out and power it externally with either a bipolar bench supply or 2 9V batteries (look up how they do that with guitar pedals).  

[SimplifiedMatrix.zip](https://github.com/Architeuthis-Flux/Jumperless/files/13847211/SimplifiedMatrix.zip)

vinilsaji2003, 2024-01-06:
Is the above shown schematic the same you have used with v0.0.0.2,

Architeuthis-Flux, 2024-01-06:
Not really, I just wired that up for you just now. But it's probably sort of similar.

vinilsaji2003, 2024-01-06:
Thankyou for your effort,by any chance do you have the schematic of v0.0.0.2,it is not present in 'breadWare',Sorry to bother you again

vinilsaji2003, 2024-01-06:
By my understanding of v0.0.0.2 we select the chips also,so we should make sure that no interconnections have been made,is that right

Architeuthis-Flux, 2024-01-06:
No, I just did one that in my head, but you can change the schematic I sent by putting the numbered labels on the side with 16 connections and the AB0-7 on the side with 8 (Y0-7).

And here's the [datasheet for the CH446Q](https://www.wch-ic.com/downloads/CH446DS1_PDF.html) which will explain the Chip Select

vinilsaji2003, 2024-01-06:
I mean we have to give the instructions to the microcontroller to switch between chips a and b by ensuring that no interconnections will be made

vinilsaji2003, 2024-01-06:
Also sir,how did you start with using KiCad

vinilsaji2003, 2024-01-07:
Sir, in our main schematic from jumperless , if we need to connect row 2 of breadboard to row 3,we need to connect y1 to y2,but if we need to do that we should connect y1 to any point from x0...x15 and then connect that point to y2, wouldn't that mean that we are actually making use of point from x0-x15 that we might need to connect between chips in the future or is that how we approach it-- do we use a point which is less likely to be used in the future, like we might be able to use x15 since it is less likely that we connect a point from chip a to chip h, But does it not leave us with one pathway less for inter-chip connections, should we use one more chip for all the intra chip connections

Architeuthis-Flux, 2024-01-07:
Yes, how that's dealt with in Jumperless's code is that it sets those X lines to -2, which signifies "I don't care which line", then when it's finished routing everything else, it goes through and sets them to the next available line. That way it doesn't interfere with a connection it needs to make. 

And when you add a single connection on the Jumperless, it clears all the paths and recalculates every path again. It helps a lot to not have to worry about what was connected and disconnected previously.

vinilsaji2003, 2024-01-07:
Thankyou

vinilsaji2003, 2024-01-07:
Is there a source from which i can download your footprint libraries----JumperlessFootprints:LQFP44_Tight
                                                                                                                   JumperlessFootprints:DIP-8_W7.62mm

Architeuthis-Flux, 2024-01-07:
Just use the standard LQFP44 footprint, it will be much easier to solder, I just had to make mine smaller so they'd fit on a really dense board. Same thing with the DIP-8, that's the standard size you'd find anywhere.

closed 2024-04-05 by Architeuthis-Flux