Architeuthis-Flux/Jumperless #23: Wokwi Bridge (Silicon) failure at decoder.py

jknudstrup, 2024-01-07:
I have tried to run the bridge on two different silicon Macs and both crash on the same error:

```
Running project Foo

Do you want to enable Auto-flashing the Arduino from Wokwi? y/n

y

Save your Wokwi project to update the Jumperless

Enter 'menu' for Bridge App menu

Traceback (most recent call last):
  File "JumperlessWokwiBridge.py", line 743, in <module>
  File "json/__init__.py", line 346, in loads
  File "json/decoder.py", line 337, in decode
  File "json/decoder.py", line 355, in raw_decode
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
[18018] Failed to execute script 'JumperlessWokwiBridge' due to unhandled exception: Expecting value: line 1 column 1 (char 0)
[18018] Traceback:
Traceback (most recent call last):
  File "JumperlessWokwiBridge.py", line 743, in <module>
  File "json/__init__.py", line 346, in loads
  File "json/decoder.py", line 337, in decode
  File "json/decoder.py", line 355, in raw_decode
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

Saving session...completed.

[Process completed]

```

jknudstrup, 2024-01-07:
This is on the latest app, flashed with the latest firmware

Architeuthis-Flux, 2024-01-07:
Huh, I'll look into that and get it figured out. 
In the meantime, could you run

`pip3 list`

in Terminal and paste what it returns.

Architeuthis-Flux, 2024-01-07:
Ah shit, Wokwi added a README.md to their projects, which moved the index where the diagram.json is stored. I'll come up with a more robust fix tomorrow (where it will work with both old and new projects) but for now, let me know if this works. 

[JumperlessWokwiBridgeAppleSiliconTempFix.zip](https://github.com/Architeuthis-Flux/Jumperless/files/13852264/JumperlessWokwiBridgeAppleSiliconTempFix.zip)

The guy who makes Wokwi even warned me about this when I first made this app, so it's totally on me for not actually searching for the right field and just assuming it would be the second one.

Architeuthis-Flux, 2024-01-07:
Hey, here's a more robust version. 
[JumperlessWokwiBridgeAppleSilicon.zip](https://github.com/Architeuthis-Flux/Jumperless/files/13854323/JumperlessWokwiBridgeAppleSilicon.zip)

Let me know if that works, I packaged it differently to allow me to sign it with apple.

closed 2024-01-07 by Architeuthis-Flux

jknudstrup, 2024-01-07:
They both work 👍 though the robustified version still requires manual approval in Privacy & Security (though that might just be a me problem)

Architeuthis-Flux, 2024-01-07:
Ah well, it was worth a shot at signing, I'll try again next time something breaks. The problem is that I've included the entire arduino-cli into the app, so who knows how it handles that. 

Thanks for opening the issue and also confirming that it works.