Architeuthis-Flux/Jumperless #30: In probe mode, apparently removed connections persist after committing

AdamSLevy, 2024-02-17:
In probe mode, connections can be "removed" by repeatedly probing the connected rails in sequence two times, which causes the LEDs for those rails to turn off. However the rails remain connected and their LEDs turn back on after committing the changes. 

It's not clear if removing connections is supported or if the LED turning off is itself a bug.

It would be very useful to be able to remove connections or discard changes and exit probe mode without committing.

Architeuthis-Flux, 2024-02-17:
Good call, I'll make a function to remove nodes from the stored nodefile. Now that there's a long press function for the probe button, maybe a long press would put you in delete connections mode or something? 

Let me know what would seem intuitive to you.

Architeuthis-Flux, 2024-02-19:
Hey this should be fixed in the latest release, 1.3.5. Feel free to reopen the issue if it's still not working.

closed 2024-02-19 by Architeuthis-Flux