Architeuthis-Flux/Jumperless #32: SSL Exception

Toube94, 2024-06-25:
When i try to open Jumperless.exe i recieve the following error:

[image]

I try to use the software for the board on a win pc.
The error only appears if i connected with the company network which is using a firewall. If i use a mobile hotspot the software works as expected

Architeuthis-Flux, 2024-06-25:
Oh that's interesting, thanks for posting the traceback!

I may have already fixed that issue by adding this line:
[image]

But I'm not actually sure if the Windows version of the packaged has that change included or not. So that could either be the problem or the fix. Schrödinger's bug. 

I get back from Portland tonight so I'll have access to a PC and can package the .exe and you can let me know if that fixes things. 

If you happen to have some spare time behind the firewall, it would give me a head start if you ran the Python script from [Releases](https://github.com/Architeuthis-Flux/Jumperless/releases/latest) and let me know if the same thing happens with and without that line commented out. But it's not a big deal either way, I'll mess with it until I get something that works, after I figure out how to set up a firewall and a dummy company network, ha.

Toube94, 2024-06-26:
When i run the .py file then i recieve the following error:
`Traceback (most recent call last):
  File "...\JumperlessWokwiBridge.py", line 6, in <module>
    from bs4 import BeautifulSoup
ModuleNotFoundError: No module named '`

Architeuthis-Flux, 2024-06-26:
That particular issue comes from not having the required modules installed. The requirements.txt file in the release should list all the modules needed to run it. In a way that can be read by whatever Python interpreter you're using, or just do >> `pip install beautifulsoup4` and then do that for the next "no module found" error that comes up. But I will not subject you to the nightmare that is Python's dependency management / pip. I wouldn't wish having to navigate that mess on my worst enemy. Thanks for giving it a shot though.

I just got home a couple hours ago so I can look into this tomorrow morning and hopefully push a fix either tomorrow or the next day. I hope your hotspot data can hold out until then.

Toube94, 2024-07-01:
Have you already been able to locate the error?

Architeuthis-Flux, 2024-07-01:
Hey @Toube94, so I've repackaged the windows app with some fixes. It's now in the [latest releases](https://github.com/Architeuthis-Flux/Jumperless/releases/latest)

Since I can't really reproduce your exact company firewall setup, I'm relying on you to let me know whether or not this works. If not I'll keep trying stuff until we get this working.

Toube94, 2024-07-02:
I still recieve ssl fail :(
[image]

Architeuthis-Flux, 2024-07-02:
Thanks again for the traceback, it's very helpful. 

It looks like everything is doing what it's supposed to do, I just need to sign this app properly to appease the firewall. I'll get on that today.

And yeah it will work fine without internet, all those menus are just being spit out from the Jumperless itself as ASCII text. Just use any Terminal Emulator, like [PuTTY](https://www.putty.org/), [coolterm](https://freeware.the-meiers.org/), [RealTerm](https://learn.sparkfun.com/tutorials/terminal-basics/real-term-windows) or whatever. There should also be a built-in command for that in PowerShell or Command Prompt to do that, (I only know on macOS and maybe Linux it's >>screen, but it will be different on Windows).

Toube94, 2024-07-08:
Maybe I misunderstood you in your last comment, but I tried to open Jumperless.exe without internet and I still get an error.

[image]

Architeuthis-Flux, 2024-07-08:
Okay, I've reworked the app to check whether you're connected to the internet sooner so it won't try to connect to Wokwi and crash. 
Just grab the app again from the [latest release](https://github.com/Architeuthis-Flux/Jumperless/releases/latest) and let me know if that works for you.

Toube94, 2024-07-10:
it works now. Thank you so much. Now I can also start the program without a hotspot because it simply does not recognize an Internet connection due to the SSL exception :)

Last question:
is it possible to communicate without jumperless.exe to the board? For example serial console or something like that

Architeuthis-Flux, 2024-07-10:
Awesome.

Yeah it will work with any serial console. With the internet off, Jumperless.exe is literally just doing that. I use the one built into VScode a lot while I'm writing code.

> And yeah it will work fine without internet, all those menus are just being spit out from the Jumperless itself as ASCII text. Just use any Terminal Emulator, like [PuTTY](https://www.putty.org/), [...]

closed 2024-09-21 by Architeuthis-Flux