Architeuthis-Flux/Jumperless #33: Arduino Nano ESP32 netlist generation problem

fpetzold, 2024-07-02:
I am relying to connect a TFT display to an Arduino Nano ESP32 board on the Jumperless. Ir works fine when I (wrongly) select an ordinary Arduino Nano as the controller in Wokwi but one connection does not get generated in the setlist if I use the board-arduino-nano-esp32.

Working project: [https://wokwi.com/projects/401031345191044097](https://wokwi.com/projects/401031345191044097)
Failing project: [https://wokwi.com/projects/400405843811118081](https://wokwi.com/projects/400405843811118081)

Working netlist:

```
netlist (Working, arduino)

Index	Name		Number	Color		Nodes
0	Empty Net	127	0x0		EMPTY_NET		{0-0}
1	GND		1	0x1C00		GND,14			{14-GND}
2	+5V		2	0x1C0702	5V,8,15			{5V-8,15-5V}
3	+3.3V		3	0x1C0107	3V3			{0-0}
4	DAC 0		4	0x231111	DAC_0			{0-0}
5	DAC 1		5	0x230913	DAC_1			{0-0}
6	I Sense +	6	0x232323	I_POS			{0-0}
7	I Sense -	7	0x232323	I_NEG			{0-0}

Index	Name		Number	Color		Nodes
8	Net 8		8	0x201B00	D13,9			{D13-9}
9	Net 9		9	0x102000	D11,10			{D11-10}
10	Net 10		10	0x2013		13,D8			{13-D8}
11	Net 11		11	0x1520		12,D9			{12-D9}
12	Net 12		12	0x20		11,D10			{11-D10}
```

Failing netlist:
```
netlist (failing, ESP32)

Index	Name		Number	Color		Nodes
0	Empty Net	127	0x0		EMPTY_NET		{0-0}
1	GND		1	0x1C00		GND,8,14		{8-GND,14-GND}
2	+5V		2	0x1C0702	5V,15			{15-5V}
3	+3.3V		3	0x1C0107	3V3			{0-0}
4	DAC 0		4	0x231111	DAC_0			{0-0}
5	DAC 1		5	0x230913	DAC_1			{0-0}
6	I Sense +	6	0x232323	I_POS			{0-0}
7	I Sense -	7	0x232323	I_NEG			{0-0}

Index	Name		Number	Color		Nodes
8	Net 8		8	0x201B00	9,D13			{9-D13}
9	Net 9		9	0x102000	10,D11			{10-D11}
10	Net 10		10	0x2013		11,D10			{11-D10}
11	Net 11		11	0x1520		12,D9			{12-D9}
12	Net 12		12	0x20		13,D8			{13-D8}
```

Note the difference in index 2, the 5V is also connected to "8" in the Arduino case.

I had a look at the python code but could not figure out where the additional connection for 5V is coming from. I suppose this connection is the one that makes the difference.

Architeuthis-Flux, 2024-07-02:
It looks like you misplaced one of the jumpers in Wokwi:

Working project:
[image]

Non working project:
[image]

Let me know if that fixes it.

fpetzold, 2024-07-02:
I feel extremely stupid now... Yes, that fixed it.

closed 2024-07-02 by fpetzold

Architeuthis-Flux, 2024-07-02:
Don't feel bad, I've spent hours trying to fix what I thought was a routing bug that ended up being exactly this.