Architeuthis-Flux/Jumperless #34: Jumperless Gets Hot

burnt-past, 2024-07-10:
During operation, the CH446Q chips located on the rear side of the PCB become heated to temperatures exceeding 200°F. As they reach this temperature, the outputs appear to cease functioning correctly. Initially, the grounding (GND) and power (VCC) connections are affected, followed by interruptions in pin-to-pin connectivity.

Architeuthis-Flux, 2024-07-10:
Yeah that shouldn't be happening. First, let me know your name so I can ship you out a fresh one and parts to fix the old one. 

It might not be permanently damaged so let's do some troubleshooting.

- Was there external power applied somewhere on the breadboard before you plugged it in? Or just send a pic of your setup.

- Is one particular chip much hotter than the others?

- Roughly when was the firmware last updated?

- With the Rail Selection Switch at the top (+-8V) what voltage do the rails read when it's first plugged in? If it's connected through a USB hub, does that change when you use something else to power it (like USB charger plugged into the wall)?

I'll try to think of more possible causes. But these should get us started in the right direction.

burnt-past, 2024-07-10:
Hey!

My name is Alex Woolum. But I ordered the jumperless through JLC at the time since you didn't have stock in your shop a awhile ago. I've always noticed it getting hot since I've had it though.

There was an external power supply attached to the jumperless in this instance. But I turned it off during power cycles of the jumperless. I was using optocouplers with an isolated power supply.

At Boot the rails are defaulting to 3.3 volts even with the switch in the 5 volt position. 

I'm on the most current firmware.

Thanks for being so prompt on your response!

Architeuthis-Flux, 2024-07-10:
Oh hey Alex,

That almost sounds like some part on the order was flipped, maybe even the Rail Supply Switch. JLCPCB has a tendency to just randomly rotate components. Someone in the Discord ordered 10 and even after rotating the parts correctly in the order, JLC flipped all the LEDs. So brutal.

From the time you ordered the clips, you have a Rev 2 with the old style clips, so it's possible something in the firmware is messing with it. My first thought is that I might have broke the code that sets the DACs to 0V at startup, and that's drawing too much current from the +-9V supply so it's sagging to a lower voltage and possibly causing this.

Anyway, have you moved since October? It's killing me to have people using my less-polished work, so I'm going to send you a fresh Rev 3 and then you can compare the orientations of the parts and make sure they're the same (or just use that and put the Rev 2 in a drawer, ha.)

If you want, send me close up photos of the front and back of the board and I might be able to spot the issue.

Architeuthis-Flux, 2024-07-11:
[image]
[image]

btw here's the correct orientation of the switch (note the little detent thing on one side), even though it seems weird it would work at all with it was upside down.

burnt-past, 2024-07-11:
Hey, Still I'm still at the same address!

Here is the thermal on the board. I swapped those out, and they ended up frying again.

[image]

Sorry for the potato photos. I need to figure out what's going on with my phone camera.  As far as I can tell, the components are all in the correct orientation.

[image]
[image]

[image]
[image]
[image]

burnt-past, 2024-07-11:
I assume it relates to the reset lines since the outputs get locked at 2V. But I'm not 100% sure; I don't have any experience with these ICs.

I'll have to depopulate those ICs next week, as I don't have time today. I'll let you know what the outcome is.

Thanks for sending out that replacement btw!

Here's the top view:

[image]

closed 2024-09-21 by Architeuthis-Flux